tasmota/tasmotizer #116: Seems to crash on Windows 11

vansen, 2021-11-05:
Hi, I recently upgraded to Windows 11. On the same device with Windows 10 flashing worked fine. Tasmotizer opens, connects to COM6 to Wemos D1 mini. After Tasmotize button is pressed, the next screen opens for ~2 seconds, then complete application just exits without any progress changed.
I assume it has to do with Win 11 but maybe someone else is able to verify?
Regards

jziolkowski, 2021-11-05:
run tasmotizer from cmd.exe and there should be a crash report, paste it here

adizag, 2022-02-17:
@vansen run as administrator

kruebsam, 2023-08-24:
Using Windows 11 the serial port gets identified (and is for sure not in use by any other app), but seems not to be really accessible by tasmotizer.
I did start tasmotizer from the command shell. unfortunately there is no more log-data.
I suspect the other user was experiencing the very same issue. That´s why I do post my readings in here.
[image]

paolosaudin, 2023-08-29:
The same here, on Windows11 does not work, on Windows10, with same harware it works.
Regards

coenier, 2023-09-05:
For me it was to run the app as administor with compatibility mode set to windows 8
[image]

paolosaudin, 2023-09-09:
Thank, I tried with  no success, unfortunately same error...

Link
[image]

Tasmotizer
[image]

Driver
[image]

Win version
[image]

paolosaudin, 2023-10-24:
> salve, sto riscontrando lo stesso problema avete risolto voi?

Ho usato il driver trovato qui (https://www.pololu.com/docs/0J7/all#2) ed
ora funziona su Win11
Paolo

fritserasmus, 2024-01-18:
I am still unable to run Tasmotizer on win 11 pc.

It seems like Windows built-in "windows-security" does not like it - Windows thinks it is a threat and deletes the file.

Any suggestions to get around this

jziolkowski, 2024-01-19:
@fritserasmus unfortunately there's very little I can do about it. Basically it boils down to me paying around 100€ per year for certificate that will make Windows Defender (and other AV software) happy.

Solution? Install python 3.8 in your system, and install tdmgr from source or PyPI package. Not much else to do.

Some days ago I've released a most up-to-date dev build here: https://pypi.org/project/tdmgr/0.2.14.dev35/

Lucacastro21, 2024-01-19:
sono riuscito a flashare tasmota su windows 11 cambiando un parametro nel driver della porta COM...
vai su gestione attività vai nell driver della tua porta COM poi su proprietà  e su port setting poi su advanced... e attiva 
Enabling the Serial port Enumerator (SerEnum). 
io facendo cosi non ho Piu problemi...

jziolkowski, 2024-01-19:
> I am still unable to run Tasmotizer on win 11 pc.
> 
> It seems like Windows built-in "windows-security" does not like it - Windows thinks it is a threat and deletes the file.
> 
> Any suggestions to get around this

I can only recommend installng using python package. Windows Defender doesn't like python exes, and I would have to pay around 100€ per year from my own pocket for a certificate to make it happy.

bartsnijder, 2024-03-09:
If you run the program as administrator it works for me

Dane8991, 2024-04-12:
I´ve got the same problme. Is there a solution for windows 11?

Dane8991, 2024-04-15:
Hey guys,

found a alternative solution which works fine for me. 
Use the tasmotizer online version found on https://tasmota.github.io/install/

paolosaudin, 2024-04-25:
Is working here as well, thanks!