tasmota/tasmotizer #136: error: metadata-generation-failed

Metti93, 2023-01-14:
Hello together

I've been trying to install Tasmotizer on my Macbook Air with macOS 13 for a few days now.
I followed the instructions with option 2:

> Option 2: Install from PyPI using pip:
> Upgrade pip and wheel packages: `pip3 install --upgrade pip wheel`
> Install tasmotizer: [pip3 install tasmotizer](url)
> Then simply run `tasmotizer.py` or `python3 -m tasmotizer.py` from the shell/command line

but I always get the same error:
What do i wrong?

```
USER@Air % pip3 install tasmotizer
Defaulting to user installation because normal site-packages is not writeable
Collecting tasmotizer
  Using cached tasmotizer-1.2.1-py3-none-any.whl (196 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting PyQt5>=5.10
  Using cached PyQt5-5.15.7.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [29 lines of output]
      Traceback (most recent call last):
        File "/Library/Python/3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 144, in prepare_metadata_for_build_wheel
          hook = backend.prepare_metadata_for_build_wheel
      AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'

      During handling of the above exception, another exception occurred:

      Traceback (most recent call last):
        File "/Library/Python/3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 351, in <module>
          main()
        File "/Library/Python/3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 333, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/Library/Python/3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 148, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
        File "/private/var/folders/dt/9y19v2kx4h9bwmyp19nfrsb40000gn/T/pip-build-env-6ppw8dti/overlay/lib/python3.8/site-packages/sipbuild/api.py", line 46, in build_wheel
          project = AbstractProject.bootstrap('wheel',
        File "/private/var/folders/dt/9y19v2kx4h9bwmyp19nfrsb40000gn/T/pip-build-env-6ppw8dti/overlay/lib/python3.8/site-packages/sipbuild/abstract_project.py", line 87, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "/private/var/folders/dt/9y19v2kx4h9bwmyp19nfrsb40000gn/T/pip-build-env-6ppw8dti/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 585, in setup
          self.apply_user_defaults(tool)
        File "project.py", line 69, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/dt/9y19v2kx4h9bwmyp19nfrsb40000gn/T/pip-build-env-6ppw8dti/overlay/lib/python3.8/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/dt/9y19v2kx4h9bwmyp19nfrsb40000gn/T/pip-build-env-6ppw8dti/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 236, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "/private/var/folders/dt/9y19v2kx4h9bwmyp19nfrsb40000gn/T/pip-build-env-6ppw8dti/overlay/lib/python3.8/site-packages/pyqtbuild/builder.py", line 69, in apply_user_defaults
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
USER@Air %
```

McrESP, 2023-02-25:
I am having the same problem with Mac-Mini M2 Pro

chuckhassett@Chucks-Mac-mini ~ % pip3 install tasmotizer 
Collecting tasmotizer
  Using cached tasmotizer-1.2.1-py3-none-any.whl (196 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting PyQt5>=5.10
  Using cached PyQt5-5.15.9.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      Traceback (most recent call last):
        File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/sy/bnjhbhxd56725zj688vnqvx00000gn/T/pip-build-env-bnumnarm/overlay/lib/python3.11/site-packages/sipbuild/api.py", line 46, in build_wheel
          project = AbstractProject.bootstrap('wheel',
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/sy/bnjhbhxd56725zj688vnqvx00000gn/T/pip-build-env-bnumnarm/overlay/lib/python3.11/site-packages/sipbuild/abstract_project.py", line 87, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "/private/var/folders/sy/bnjhbhxd56725zj688vnqvx00000gn/T/pip-build-env-bnumnarm/overlay/lib/python3.11/site-packages/sipbuild/project.py", line 585, in setup
          self.apply_user_defaults(tool)
        File "/private/var/folders/sy/bnjhbhxd56725zj688vnqvx00000gn/T/pip-install-sti8fsfx/pyqt5_2d741dc9f6bc4c3cb755de8ba2269893/project.py", line 68, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/sy/bnjhbhxd56725zj688vnqvx00000gn/T/pip-build-env-bnumnarm/overlay/lib/python3.11/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/sy/bnjhbhxd56725zj688vnqvx00000gn/T/pip-build-env-bnumnarm/overlay/lib/python3.11/site-packages/sipbuild/project.py", line 236, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "/private/var/folders/sy/bnjhbhxd56725zj688vnqvx00000gn/T/pip-build-env-bnumnarm/overlay/lib/python3.11/site-packages/pyqtbuild/builder.py", line 69, in apply_user_defaults
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
chuckhassett@Chucks-Mac-mini ~ %

boonya, 2023-03-01:
Same on **Mackbook Pro 13" M1**

```
$ pip3 install tasmotizer
Collecting tasmotizer
  Using cached tasmotizer-1.2.1-py3-none-any.whl (196 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting PyQt5>=5.10
  Using cached PyQt5-5.15.9.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      Traceback (most recent call last):
        File "/opt/homebrew/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/opt/homebrew/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/opt/homebrew/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/s3/n_xxz6bj3z77s690s8nx3kc00000gq/T/pip-build-env-a_mbuhcv/overlay/lib/python3.11/site-packages/sipbuild/api.py", line 46, in build_wheel
          project = AbstractProject.bootstrap('wheel',
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/s3/n_xxz6bj3z77s690s8nx3kc00000gq/T/pip-build-env-a_mbuhcv/overlay/lib/python3.11/site-packages/sipbuild/abstract_project.py", line 87, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "/private/var/folders/s3/n_xxz6bj3z77s690s8nx3kc00000gq/T/pip-build-env-a_mbuhcv/overlay/lib/python3.11/site-packages/sipbuild/project.py", line 585, in setup
          self.apply_user_defaults(tool)
        File "/private/var/folders/s3/n_xxz6bj3z77s690s8nx3kc00000gq/T/pip-install-9nhv0ke2/pyqt5_9951acb1b9424c2aafc81c802cdf492a/project.py", line 68, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/s3/n_xxz6bj3z77s690s8nx3kc00000gq/T/pip-build-env-a_mbuhcv/overlay/lib/python3.11/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/s3/n_xxz6bj3z77s690s8nx3kc00000gq/T/pip-build-env-a_mbuhcv/overlay/lib/python3.11/site-packages/sipbuild/project.py", line 236, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "/private/var/folders/s3/n_xxz6bj3z77s690s8nx3kc00000gq/T/pip-build-env-a_mbuhcv/overlay/lib/python3.11/site-packages/pyqtbuild/builder.py", line 69, in apply_user_defaults
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

tomekc, 2023-04-16:
Hey,
I've had the same problem but found solution:

 1. Install pyqt from homebrew `brew install pyqt@5`
 2. `PATH="/opt/homebrew/opt/qt5/bin:$PATH" pip3 install tasmotizer`