tasmota/tasmotizer #141: Online Flash Tool does not install the latest Release

gzi01, 2024-01-26:
I used the tool at https://softwarecrash.github.io/Daly2MQTT/espflashtool/ to flash a Wemos and got release 2.8.3 instead of 2.12.0 .

I would have expected the latter or at least options to slect a release.

By the way: Tried to use Tasmotizer to flash 2.12.0 but failed. Are there any descriptions how to do a firmware update with Tasmotizer?

jziolkowski, 2024-01-26:
Tasmotizer uses flags specific for flashing tasmota firmware. Other firmwares are not supported.

I suggest you try esptool manually but you still need to follow the instructions for the firmware you are trying to use.

gzi01, 2024-01-26:
As I am not familiar with esptool , can you please provide me a hint where to find Info about flashing DALY2MQTT binaries to a Wemos D1 mini?

jziolkowski, 2024-01-26:
No idea. Never heard of this software before. My Google is as good as yours.

gzi01, 2024-01-29:
just found out that this was misleaded. sorry. closing.

gzi01, 2024-01-29:
done

closed 2024-01-30 by Jason2866